betaflight/config #360: BETAFPVF435 target

NobodyExceptHim, 2024-03-22:
### Is your feature request related to a problem? Please describe

There is a betafpvf435 target inside firmware flasher according to betaflight 4.5 but there are no builds to select from.

### Describe the solution you'd like

Know if its a config issue or if there is no plan to add support to this board

### Describe alternatives you've considered

ATbetaflight but it is too old

haslinghuis, 2024-03-22:
Get configurator from https://github.com/betaflight/betaflight-configurator/releases

[image]

NobodyExceptHim, 2024-03-22:
[image]
I have the newest version as you can see in the top left corner :(

NobodyExceptHim, 2024-03-22:
I'm terribly sorry. Release candidates ;)... Thank you!

closed 2024-03-22 by haslinghuis

NobodyExceptHim, 2024-03-22:
@haslinghuis Could I also ask you (cause i truly don't know how to check it) if this version allows built in elrs v3 with this target?

haslinghuis, 2024-03-22:
If you mean ERLS SPI - the firmware supports V3.

You can ask question on Betaflight [Discord Server](https://discord.com/invite/gV4XSq3fus)