betaflight/config #440: CLNE/FREEAXISF4: no RGB LED, gyro/accel, motors working ootb

sixtyfive, 2024-05-29:
I recently bought another one of these [FreeAxis F4 25A](https://de.aliexpress.com/item/1005005748337742.html) AIOs since I liked the old one that ended up in the ocean. I still remember it was a little difficult to figure out. Currently I'm tinkering with the new one to get all peripherals back to work.

So far I have the mnotor and the gyro/accel working.

Unfortunately the RGB LED is proving a little more difficult. According to the BF target created last year, it should be on A08.

Short from desoldering it, how would one usually reverse engineer this?

nerdCopter, 2024-05-31:
- you did include LED_STRIP in the could build option?
- did you test a `core-only` build? 
- can you post a build-log(s)?

i.e. troubleshooting if this is a build issue, or hardware defines issue. (i'm not personally knowledgeable on many of the define addressing/pins)

sixtyfive, 2024-05-31:
Yes, `USE_LED_STRIP`, was included. I'm happy to test a `core-only` build, but was under the impression that that wouldn't have LED strip support.

Will post build logs once I'm back home.

nerdCopter, 2024-05-31:
> [...] was under the impression that that wouldn't have LED strip support.

i had to test myself, seems it is included. 
[image]

sixtyfive, 2024-06-01:
Okay, so this is a little weird. On the `core-only` build, `A08` is already taken by `LED 1`. So after I do `resource LED 1 none; save`, then `resource LED_STRIP 1 A08; save`, it's back to `LED 1 A08`.

Back to fresh cloud build, where it already has `resource LED_STRIP 1 A08` out of the box. Didn't change anything else for now, only set up 1 single LED. This is what that should look like, right?

[image]

sixtyfive, 2024-06-02:
Today, tried soldering a NeoPixel to the LED pin thinking perhaps the builtin RGB LED is bust. But still no light. Tried soldering the NeoPixel to the SmartAudio pin instead and reconfiguring resources to use `A00` for `LED_STRIP`. Still nothing. I'm mildly convinced now that it is not a hardware thing but something with the build.

nerdCopter, 2024-06-03:
i don't know enough about LED_STRIP versus/and onboard LEDs. someone else will have to review.

sixtyfive, 2024-06-05:
@nerdCopter the riddle is solved: it turns out, once you actually _power_ the LED, lo and behold, the LED will light up... 🤦‍♂️

Anyhow, that only leaves for all the `#define`s to be correct. I think I might be able to figure that out and then PR it. Closing this, and thank you for your patience and help!

closed 2024-06-05 by sixtyfive