betaflight/config #444: Applying 4.4/10.9 diff via CLI or loading saved backup via Presets in 4.5/10.10 fails to enable and always disables Serial RX option and OSD option

Darkmann12, 2024-05-31:
### Describe the bug

When applying a diff taken from a (fully working) quad that is on 4.4, when the FC has been freshly flashed to 4.5, disables the OSD option in the Configuration tab (even though the OSD elements and their positions load correctly once the option is manually enabled again), and also disables Serial RX on the correct UART in the Ports tab. Once this has been manually rectified, reapplying the configuration/diff/backup preset disables the options again, despite not having anything within their text content that specifically performs this disabling (as per a couple members V22 and Vit). This results in the outcome of loading a saved backup not resulting in a usable craft.

### To Reproduce

Take a diff of a configuration, or save a backup, from a quad on 4.4 firmware (that has (hd) OSD enabled and one serial rx enabled), flash to latest firmware with full chip erase, and apply the diff/backup.

### Expected behavior

all options should transfer over - serial RX should be enabled on the UART that it was, and OSD should be enabled

### Configurator version

10.10.0 (c97deaf)

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.5.0 May 31 2024 / 17:34:57 (c155f5830) MSP API: 1.46
# config rev: 24cb4e0

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF411
manufacturer_id DIAT
mcu_id 0045001c3439510436383737
signature 

# feature
feature TELEMETRY
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 1 131073 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 0 1600 2100 0 0
aux 1 13 2 900 1300 0 0
aux 2 35 1 1625 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -34,-26,45,1
set min_check = 1025
set max_check = 1950
set rc_smoothing_auto_factor = 120
set dshot_bidir = ON
set motor_poles = 12
set vbat_max_cell_voltage = 440
set ibata_scale = 165
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set report_cell_voltage = ON
set osd_rssi_dbm_alarm = -95
set osd_tim2 = 1
set osd_link_quality_pos = 3627
set osd_link_tx_power_pos = 3175
set osd_rssi_dbm_pos = 3595
set osd_tim_2_pos = 2083
set osd_throttle_pos = 3144
set osd_current_pos = 2114
set osd_mah_drawn_pos = 2601
set osd_warnings_pos = 14868
set osd_avg_cell_voltage_pos = 2569
set osd_flip_arrow_pos = 2394
set osd_framerate_hz = 10
set debug_mode = GYRO_SCALED
set cpu_overclock = 120MHZ

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set dyn_idle_min_rpm = 40

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Add any other context about the problem that you think might be relevant here

using HD osd

haslinghuis, 2024-05-31:
Only include OSD_HD not OSD_SD while flashing. When above PR is merged UART 1 should default to SERIALRX

Darkmann12, 2024-05-31:
> Only include OSD_HD not OSD_SD while flashing. When above PR is merged UART 1 should default to SERIALRX

This was done in both instances on the 4.5 side. As for the 4.4 side I was running into that problem with selecting HD OSD not working at all and so if only had HD OSD at compile time, was soft locked out of HD OSD.

PR #445 (2024-06-29): Add SERIALRX_UART for MAMBAF411
- fixes #444
closed 2024-06-29 by haslinghuis