betaflight/config #606: New ID for FLYTEX LTD

MatviiG, 2024-11-07:
Hi,

could you please add an ID for FLYTEX LTD
website: https://www.flytex.pro/
Targets: FLYTEX405 ; FLYTEX722

BR,
Matvii

haslinghuis, 2024-11-07:
@sugaarK

MatviiG, 2024-12-04:
> Checkout https://betaflight.com/docs/development/manufacturer/requirements-for-submission-of-targets
> 
> @sugaarK

Hi @haslinghuis sorry for bothering you. I checked documentation You pointed - it says I need ID which I'm asking here, Information about company already provided, targets will be provided just after ID obtained. Or we should provide hardware design for review first?

nerdCopter, 2024-12-04:
you simply need to choose a 4-cahracter ID and pull-request to https://github.com/betaflight/config/blob/master/Manufacturers.md in the same format.

maybe `FLYT`, maybe `FTEX`

example (in the proper sort order):
```
|FLYT|FLYTEX LTD|https://www.flytex.pro/|
```
Note, there is a UK company named FLYTEX LTD (oil company)

cvetaevvitaliy, 2024-12-04:
@MatviiG You make clones on StingBee electronics

cvetaevvitaliy, 2024-12-04:
@sugaarK Please pay attention.

MatviiG, 2024-12-04:
> @MatviiG You make clones on StingBee electronics
In case we make clones - why we need our own id for configuration?

cvetaevvitaliy, 2024-12-04:
> > @MatviiG You make clones on StingBee electronics
> > @cvetaevvitaliy In case we make clones - why we need our own id for configuration?

hm, Say hello to Andrew
Stealing electronics drawings from a factory and appropriating them for yourself is not very nice.

sugaarK, 2024-12-05:
this isn't going any where until the issue of cloning is addressed

MatviiG, 2024-12-10:
@cvetaevvitaliy  So. as You talked yesterday with us about FC we are going to produce - are there still objections from your side about it? Is it still is blocked?

cvetaevvitaliy, 2024-12-10:
> @cvetaevvitaliy  So. as You talked yesterday with us about FC we are going to produce - are there still objections from your side about it? Is it still is blocked?

No, we thought about the option of cooperation, nothing more. 
Need time for final.

PR #627 (2024-12-19): Add FLTE manufacturer
- closes #606
- awaiting a target
- draft update for target inclusion effective immediate: https://deploy-preview-493.dev.web.betaflight.com/docs/development/manufacturer/config-target-guidance
closed 2024-12-19 by sugaarK