betaflight/config #611: cloud build error - impossible to update FC target HGLRCF722AIOV2 for betaflight 4.5.x

sghizz, 2024-11-18:
### Describe the bug

Hi, i have a hglrc specter AIO 25A. The target is HGLRCF722AIOV2. It comes with BF 4.4.x. 
It is impossible to update to 4.5 or 4.5.1. Cloud build failed. I have betaflight configurator 10.10 on windows. if i cloud build a 4.4.x version it works. if i select any other target it works with 4.5.x... Can you verify where is the problem? 

### To Reproduce

//

### Expected behavior

//

### Support ID

```txt
//
```

### Flight controller

hglrc specter aio 25A

haslinghuis, 2024-11-18:
Other possibility it is a new board and there is simply no configuration available yet.

sghizz, 2024-11-18:
Oh yes, but who cooks the firmware for this boards? I would send him a
request…

haslinghuis, 2024-12-06:
Resolved in #633

closed 2024-12-06 by haslinghuis