betaflight/config #631: HGLRCF411SX1280_15A Fails During Cloud Build

Mute916, 2024-12-06:
### Describe the bug

HGLRC Draknight 2" Toothpick comes with the HGLRCF411SX1280_15A AIO flight controller. It has build 4.4.2 on it but will not build anything above that. Including 5.0+

### To Reproduce

1. Open Firmware Flasher on Betaflight 10.10.0
2. Load HGLRCF411SX1280_15A
3. Load Firmware version 4.5.0 or above
4. Click "Load Firmware [Online]"
5. Cloud Build Status "Failed" after 10 seconds

### Expected behavior

Cloud Build (Load Firmware Online) to compile and complete the build

### Support ID

e995f326-f722-4d6a-8cb6-1305ff91e479

### Flight controller

HGLRCF411SX1280_15A, HGLRC

### Other components

ELRS EP1/EP2 RX
VTX Smart Audio 2.1
Z-H-30 4in1 ESC
STM411 FC
LED Strip

### How are the different components wired up (including port information)

Everything is all on 1 board

### Add any other context about the problem that you think might be relevant here

[HGLRC Build Log.txt](https://github.com/user-attachments/files/18041315/HGLRC.Build.Log.txt)

haslinghuis, 2024-12-06:
Try HGLRCF411SX1280

[image]

nerdCopter, 2024-12-06:
I somewhat recall a name change.  i think it is `HGLRCF411ELRS` as it seems to have the same resources.

nerdCopter, 2024-12-06:
duplicate https://github.com/betaflight/config/issues/601
i.e. 15A is `HGLRCF411ELRS`

closed 2024-12-06 by nerdCopter