betaflight/config #632: Target HGLRCF722AIOV2 doesn't work for bf 4.5

sghizz, 2024-12-06:
### Describe the bug

Hi, i have new AIO FC: HGLRC SPECTER 25A AIO. It comes with BF 4.4.x.  
The problem is that it is impossible to update to BF 4.5 or newer versions. 
The BF configurator (10.10.0) that works great with other FC like betafpv, says CLOUD BUILD FAILED see log file.

The target is HGLRCF722AIOV2 
 
Please anyone had a solution? Please try to build it on your configurator. Mine log file is: 

https://build.betaflight.com/api/builds/5df8b09b0797dbb3e3d5b7ed0fd7e1ab/log

### To Reproduce

build with 10.10.0 conf.

### Expected behavior

\\

### Support ID

```txt
\\
```

### Flight controller

hglrc specter aio 25A

haslinghuis, 2024-12-06:
Probably should be [HGLRCF722AIO](https://github.com/betaflight/config/tree/master/configs/HGLRCF722AIO)

sghizz, 2024-12-06:
> Probably should be [HGLRCF722AIO](https://github.com/betaflight/config/tree/master/configs/HGLRCF722AIO)

HI, Thank you for the response. It is another versions of this AIO. It is 40A instead of 25A of mine. Itried to flash it and some uarts doesn't work and the internal memory for blackbox.

haslinghuis, 2024-12-06:
Other possibility is `HGLRCF722AIO_X6` otherwise we do not have a target [yet].

sghizz, 2024-12-06:
> Other possibility is `HGLRCF722AIO_X6` otherwise we do not have a target [yet].

I tried also this version, other uarts doen't work. Maybe the problem is: the config.h of this target was deleted in the repo. one time it was in the repo.

see   https://github.com/betaflight/unified-targets/commit/6d307d405440c0b2a66a014f445056c2a7074988#diff-b1d8d69961c3c30a687e52645abf65e98769414c3b4d9aab6f74ba0e59421800

haslinghuis, 2024-12-06:
@sghizz V2 should be available in 15 minutes

PR #633 (2024-12-06): Migrate HGLRCF722AIOV2
- fixes #632
- please note sub optimal sharing SPI between OSD and blackbox
- missed migration in https://github.com/betaflight/unified-targets/pull/1175
- thanks to @ot0tot for improvements
closed 2024-12-06 by haslinghuis