betaflight/config #646: conflict between motors outputs and strip LED

airbotsystems, 2024-12-19:
### Describe the bug

When Led strip function is active, it works well but do not allow to access the motor page without getting an error message. Motors output are inactives.
After turning off the led strip function, motor page load normal, without error popup essga and all MOT outputs are functional.

### To Reproduce

hard because it is on a custom board, actually in dev
I suppose a conflict between dmas but not sure

### Expected behavior

Led and motors working well independentely

### Support ID

```txt
d5b49aca-2177-4cfe-9aeb-01f910e90061
```

### Flight controller

Custom H7 FC

SteveCEvans, 2024-12-19:
What’s the FC? If it’s not an F411 you can have DSHOT use TIM8.

airbotsystems, 2024-12-20:
Issue seems to come from GYRO CLK function. If i remove it from the config the issue deseaper

haslinghuis, 2024-12-20:
Closing as github is for development, not support :)
As PC9 is using TIM8 it conflicts with `dshot_bitbang`

closed 2024-12-20 by haslinghuis