betaflight/config #720: FLYWOOF405S_AIO: Update config.h

Canaill51, 2025-03-14:
FLYWOOF405S_AIO config.h file misses some features (example: PINIO).

Could you please update it according to this Id below.

Thanks in advance for your prompt support.
Cédric

See Id: "6009c2b7-aeb6-40c5-ba4a-f9dce4bbfe6a"

For reference, here is what this FC AIO contained when shipped from Flywoo:
```
# version
# Betaflight / STM32F405 (S405) 4.5.1 Feb 10 2025 / 01:43:57 (77d01ba3b) MSP API: 1.46
# config rev: 6433775
# board: manufacturer_id: FLWO, board_name: FLYWOOF405S_AIO

# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.30V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3817, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (53 x 20)
BUILD KEY: 617a6e21de02f41566486eb020885235 (4.5.1)
System Uptime: 58 seconds, Current Time: 2025-03-14T17:23:51.537+00:00
CPU:21%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT ENABLED
Arming disable flags: RXLOSS ANGLE CLI MSP DSHOT_TELEM

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608 JEDEC ID=0x00ef4017
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=0

# dump master

# version
# Betaflight / STM32F405 (S405) 4.5.1 Feb 10 2025 / 01:43:57 (77d01ba3b) MSP API: 1.46
# config rev: 6433775

# start the command batch
batch start

board_name FLYWOOF405S_AIO
manufacturer_id FLWO

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B05
resource MOTOR 6 B07
resource MOTOR 7 C09
resource MOTOR 8 NONE
resource LED_STRIP 1 A09
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
[... 1071 lines cut ...]

# timer show

Currently active Timers:
-----------------------
TIM1:
    CH2 : LED_STRIP
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

haslinghuis, 2025-03-14:
> resource PINIO 2 C08
> set pinio_box = 40,41,255,255

Only pin 2 is available here, so makes sense to configure as 
```
resource PINIO_1 C08
set pinio_box = 40, 255,255,255
```

@Canaill51 TL:DR; :) - any more examples ?

Canaill51, 2025-03-14:
> > resource PINIO 2 C08
> > set pinio_box = 40,41,255,255
> 
> Only pin 2 is available here, so makes sense to configure as
> 
> ```
> resource PINIO_1 C08
> set pinio_box = 40, 255,255,255
> ```
> 
> [@Canaill51](https://github.com/Canaill51) TL:DR; :) - any more examples ?

Thanks for your very quick feedback!
Nope, that's the only one I've been reported by some friends. They tried to enable "VTX power off" (their O4 lite) with PINIO as Flywoo explains on their website:
- One of my friend was able to, one was not able to: The difference was the first just configured when received. The second flashed it.
- I checked on config.h and did not see this PINIO, because I have this FC also :)

Let me know when you will have fixed this :)

haslinghuis, 2025-03-14:
- Hmmm. As being AIO why does the target have MOTOR 5 - 8 defines 😕 
- PIN C08 conflicts with MOTOR8

Canaill51, 2025-04-02:
Are there any updates since? Thank you :)

haslinghuis, 2025-04-02:
This Should fix it.
```
resource MOTOR 8 NONE
resource PINIO 1 C08
set pinio_box = 40,255,255,255
set pinio_config = 129,1,1,1
```