betaflight/config #778: Sub250 Nimble65 (GEPRCF411SX1280/STM32F411) unflyable on Betaflight 4.5.2 with current config due to gyro misconfiguration (4.5.1 okay)

DavidAnson, 2025-05-14:
The Sub250 Nimble65 ships (as of about 1 month ago) with <https://build.betaflight.com/api/builds/c6e8ae1147c3a84dcf4095780dd14204/log> flashed containing Betaflight 4.5.1 with config `ad0057e`. That flies fine.

The corresponding Cloud Build for Betaflight 4.5.2 is <https://build.betaflight.com/api/builds/474ed9cd32f0201161e0510e5374e625/log> with config `467f87b`. It idles okay, but if you try to take off, ZOOM TO THE MOON with audible power fluctuations on the way up. Upon inspection, the gyro is misaligned and this can be easily observed in Betaflight Configurator's Setup tab - it's about 45 degrees off so pure physical pitch or roll produce measured pitch AND roll on screen.

If you're lucky, you notice that after flashing. If not, you may have to disarm at 30 feet over concrete and crack your frame in 3 places. :|

If you're the kind of person who likes to `diff all`, you may notice that in the working configuration immediately after flashing:
```
set gyro_1_sensor_align = CW0
set gyro_1_align_yaw = 0
```

And in the broken configuration immediately after flashing:
```
set gyro_1_sensor_align = CUSTOM
set gyro_1_align_yaw = 3150
```

If you're the kind of person who likes to read GitHub commits, you may decide the problem is a COMBINATION of these two commits:

Betaflight 4.5.2: <https://github.com/betaflight/betaflight/commit/7e55f9e73c9628c7fd4309bc6039495f13db9f1d>
Betaflight config: <https://github.com/betaflight/config/commit/ac7c65c3acfef6b60c1b70fce64a381cc958feba#diff-f8cb1e76e2e76dd5c148e83ff603d845ef5d004dfa8964d3a595dbaf9ee88c20>, specifically `configs/GEPRCF411SX1280/config.h`, extra specifically:
```
+ #define GYRO_1_ALIGN ALIGN_CUSTOM
#define GYRO_1_ALIGN_YAW -45
```

I don't know enough about `GEPRCF411SX1280` to say if this configuration is wrong for all such boards, but I have confirmed experimentally that it is wrong for the Sub250 Nimble65. The fix (also confirmed experimentally) is to apply the two "working configuration" settings above after flashing Betaflight 4.5.2.

It would be ideal if this worked by default. :)

CC: @haslinghuis / @ledvinap

ot0tot, 2025-05-14:
What does this have to do with the DEPRCF411SX1280 target? This is a completely different manufacturer. Product page states it comes with this unapproved/unsupported FC: https://sub250.com/collections/fc-esc-stacks-combos/products/redfox-a1-f4-5a-4in1-aio-fc

DavidAnson, 2025-05-14:
Per the first link above, that's what came flashed on the device from the manufacturer when I received it about a month ago:
```
***** Commencing Build *****
VERSION:   4.5.1
COMMIT:    4.5.1
CONTAINER: 4.5.X
BOARD:     GEPRCF411SX1280/STM32F411
STARTED:   2024-08-01 19:42:22Z
```

I can't PROVE that since I've flashed something else since, but you can check the date above and the build location in the JSON and trust me that I didn't visit Russia to do firmware builds in the last year. :)

I can also assert the first build I link to above produces a flyable drone as-is and the second (same configuration) is only flyable with the correction I outline here. If you would like to see the output of "version" or "status" or something from the CLI for this hardware, I am happy to paste that here.

But even if you don't believe anything I'm saying, my understanding of the code/commit is that it rotates the primary gyroscope 45° for that hardware and it's hard to imagine that it's correct both before AND after the change.

nerdCopter, 2025-05-14:
large disconnect between vendor/manufacturer/betaflight-team.
- https://sub250.com/products/nimble65-whoop-quadcopter
- `SUB250F411SX1280` does not exist, so they using `GEPRCF411SX1280` apparently.

[image]

nerdCopter, 2025-05-14:
@DavidAnson , if you flash GEPRC target, i believe you can "custom define" certain parameters.  please try, i 

> - Pilot may also define values with `PARAMETER=VALUE`. e.g. `GYRO_1_ALIGN=CW180_DEG`

you may need to test/play. try `GYRO_1_ALIGN_YAW=[XXX]`

DavidAnson, 2025-05-14:
Thanks @nerdCopter! As far as I know, what I have now works and keeps me on "official" builds. I'm hoping this is something that can get changed in the repo so no one will need to worry about it ever again. But I appreciate the suggestion and will keep that idea in my back pocket just in case.

ot0tot, 2025-05-14:
That's the problem, there is no "official" build because the manufacturer has not bothered to create one. Using a different manufacturer's target is frowned upon, and certainly no changes will be made to the target because of it. 

Please take it up with the manufacturer, and point them to: https://betaflight.com/docs/development/manufacturer/requirements-for-submission-of-targets

DavidAnson, 2025-05-14:
@ot0tot, forget Sub250 and my drone for a minute. The configuration for the GEPRCF411SX1280 target’s primary gyroscope used to be 0° and now it’s 45°. Those can’t both be right. Either it was right before and it’s wrong now because it was accidentally changed by this unrelated commit OR it was wrong before and it’s right now because the fix was part of this unrelated commit. I don’t know which one it is, but this is something that can be determined by consulting the specifications for the GEPRCF411SX1280 hardware. Based on my own works/fails experience with “unsupported hardware”, I suggest that it’s wrong now.

nerdCopter, 2025-05-14:
is your board an official `GEPRCF411SX1280` by GepRC?  if it is some clone, then a clone could orient the gyro any way they desire.  
maybe add hi-res photos of your board to discern if it is official or clone.

DavidAnson, 2025-05-14:
Everybody reset your brains for a second. :) I'm going to make a set of factual statements here and I'm not going to reference unsupported hardware AT ALL. Let's just talk ONLY about the supported `GEPRCF411SX1280` target:

1. Since the beginning of history in this repo (and presumably even longer elsewhere, but for sure no later than April 27, 2023) the `GEPRCF411SX1280` target was configured as `GYRO_1_ALIGN CW0_DEG` (and also `GYRO_1_ALIGN_YAW -45`, but that didn't apply because `CW0_DEG` is not `ALIGN_CUSTOM`).
2. Every build up until December 20, 2024 (and every build since then using Betaflight 4.5.1 or earlier (for reasons I won't go into to avoid complicating things)) used that configuration with Gyro 1 at 0 degrees.
3. On December 20, 2024, the configuration of the `GEPRCF411SX1280` target was CHANGED to `GYRO_1_ALIGN ALIGN_CUSTOM` meaning the existing `GYRO_1_ALIGN_YAW -45` value newly applies.
4. Every build for this target on Betaflight 4.5.2 or later since December has Gyro 1 at -45 degrees.
5. A build that flys well at 0 degrees will not fly well at 45 degrees and vice-versa.
6. The physical hardware for this target that exists in the physical world was not affected by the configuration change in this repo.
7. At most one of these two configurations is correct for the physical hardware that exists in the world.
8. Someone who has or knows this hardware or specification can tell us the expected value.

Finally, here's an opinion: It's more likely in my mind that the `GEPRCF411SX1280` target worked for multiple years and broke recently than it is that it was broken for multiple years and only started working recently as part of a completely unrelated refactoring of the code.

Reference: <https://github.com/betaflight/config/commits/master/configs/GEPRCF411SX1280/config.h>

haslinghuis, 2025-05-14:
Hmm, had kind of the same assessment this morning.
- Does anyone have a real GEPRC board to verify?
- Was looking at change between unified target and config.

ot0tot, 2025-05-14:
Here is the FC in question:

[image]

Note the orientation arrow, and the gyro.

DavidAnson, 2025-05-14:
@ot0tot, to be clear, your thinking is that this target was unusably broken from the time it was added years ago all the way up until an unrelated code refactoring fixed it 5 months ago? And nobody noticed or reported a problem in all that time?

I'm not going to keep fighting this, but I haven't seen anyone address the question above in a way that makes sense to me.

haslinghuis, 2025-05-14:
My proposed fix was based on this knowledge:

Unified target:

```bash
set gyro_1_sensor_align = CW0
set gyro_1_align_yaw = -45
```

Config.h

```bash
#define GYRO_1_ALIGN ALIGN_CUSTOM
#define GYRO_1_ALIGN_YAW 3150
```

- As `gyro_1_sensor_align = CW0` was used before `gyro_1_align_yaw` would not apply (BUG)
- In config we use `CUSTOM` instead and `GYRO_1_ALIGN_YAW` is used