betaflight/config #839: Micoair H743 AIO V2 NO GYRO

Pancronos, 2025-07-22:
### Describe the bug

After flashing firmware 4.6 dev  no gyro detected, as well as no SD card (original from manufacturer).

### To Reproduce

flash 4.6 on new AIO Board

### Expected behavior

Should work with choice of different gyro bmi270 or bmi088

### Configurator version

11.0

### Flight controller configuration

```txt
Entering CLI Mode, type 'exit' to reboot, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32H743 (SH74) 4.6.0 Jul 17 2025 / 16:03:15 (67d9ca818) MSP API: 1.47
# config rev: 6f5fb58

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MICOAIR743
manufacturer_id MICO
mcu_id 002400203333511035353936
signature 

# feature
feature LED_STRIP
feature OSD

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Add any other context about the problem that you think might be relevant here

2025-07-17 @18:15:04 -- MultiWii API version: 1.47.0

2025-07-17 @18:15:04 -- Flight controller info, identifier: BTFL, version: 4.6.0

2025-07-17 @18:15:04 -- Running firmware released on: Jul 17 2025 16:03:15

2025-07-17 @18:15:04 -- Board: MICO/MICOAIR743(STM32H743), version: 0

2025-07-17 @18:15:04 -- Unique device ID: 0x2400203333511035353936

2025-07-17 @18:15:04 -- Build Key: aac6b67483325fa10e892a422d0e89dc

https://build.betaflight.com/api/builds/aac6b67483325fa10e892a422d0e89dc/log

https://build.betaflight.com/api/builds/aac6b67483325fa10e892a422d0e89dc/json

nerdCopter, 2025-08-04:
so essentially, they never contacted BF, nor paid for target review/inclusion?

Pancronos, 2025-08-04:
Sorry, i am a user, don't know how it works. All my interest, as it was for axsisflying AIO, is to find a valid target that i can use with latest BF functions. I' ll leave this between you and manufacturer. Feel free to close this request.